leggedrobotics/raisimGym #4: 'Segmentation fault' when run example script in docker

dytan, 2019-08-19:
Thanks for the work.
   I'd try the raisim engine for rl usage, environment is built exactly follow the given dockerfile, and to run we connect to the docker container through ssh with x-forwarding enabled. When run any example script, trigger a 'segmentation fault' error.
  After search and check, it seems to be a ogre issue, 'gl' related render plugin is not able to be loaded, run the built Orge example 'SampleBrowser' show the same error, I am not familiar with Ogre and found few related [issues](https://github.com/OGRECave/ogre/issues/810) on orge repository but not a suitable way to solve this problem, seems x forwarding not use the right gl.  Is there any way to run raisim gui through x forwarding on docker ?

the container follow the dockerfile, and runtime config :
OS: ubuntu 18.04
GPU: GTX 1080, driver 418.56, cuda 10.1

run 'python3 scripts/laikago_blind_locomation.py', output:
``` bash
[2019:08:19:02:56:24 OgreVis.hpp:295] Loading RaisimOgre Resource from: /opt/raisim_build/share/raisimOgre/cmake/../rsc
[2019:08:19:02:56:24 OgreVis.hpp:296] Loading Ogre Configuration from: /opt/raisim_build/share/raisimOgre/cmake/../ogre/
[task16227-q7qdf:09966] *** Process received signal ***
[task16227-q7qdf:09966] Signal: Segmentation fault (11)
[task16227-q7qdf:09966] Signal code: Address not mapped (1)
[task16227-q7qdf:09966] Failing at address: 0x8
[task16227-q7qdf:09966] [ 0] /lib/x86_64-linux-gnu/libc.so.6(+0x3ef20)[0x7f5c89d09f20]
[task16227-q7qdf:09966] [ 1] /usr/lib/x86_64-linux-gnu/libstdc++.so.6(_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_assignERKS4_+0x28)[0x7f5c87217058]
[task16227-q7qdf:09966] [ 2] /opt/raisim_build/lib/OGRE/RenderSystem_GL.so.1.12.1(+0xf7dec)[0x7f5b9521fdec]
[task16227-q7qdf:09966] [ 3] /opt/raisim_build/lib/OGRE/RenderSystem_GL.so.1.12.1(_ZN4Ogre14GLRenderSystemC2Ev+0x2a7)[0x7f5b951e0767]
[task16227-q7qdf:09966] [ 4] /opt/raisim_build/lib/OGRE/RenderSystem_GL.so.1.12.1(+0xaea6e)[0x7f5b951d6a6e]
[task16227-q7qdf:09966] [ 5] /opt/raisim_build/lib/libOgreMain.so.1.12.1(_ZN4Ogre4Root13installPluginEPNS_6PluginE+0xfb)[0x7f5c877d09db]
[task16227-q7qdf:09966] [ 6] /opt/raisim_build/lib/libOgreMain.so.1.12.1(_ZN4Ogre4Root10loadPluginERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x17a)[0x7f5c877d0c6a]
[task16227-q7qdf:09966] [ 7] /opt/raisim_build/lib/libOgreMain.so.1.12.1(_ZN4Ogre4Root11loadPluginsERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x545)[0x7f5c877d12d5]
[task16227-q7qdf:09966] [ 8] /opt/raisim_build/lib/libOgreMain.so.1.12.1(_ZN4Ogre4RootC1ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES8_S8_+0x135d)[0x7f5c877d33dd]
[task16227-q7qdf:09966] [ 9] /opt/raisim_build/lib/libOgreBites.so.1.12.1(_ZN9OgreBites22ApplicationContextBase10createRootEv+0x39d)[0x7f5c87bae89d]
[task16227-q7qdf:09966] [10] /opt/raisim_build/lib/libOgreBites.so.1.12.1(_ZN9OgreBites22ApplicationContextBase7initAppEv+0xf)[0x7f5c87badaef]
[task16227-q7qdf:09966] [11] /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/_raisim_gym.cpython-36m-x86_64-linux-gnu.so(+0x1dcd2)[0x7f5c88654cd2]
[task16227-q7qdf:09966] [12] /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/_raisim_gym.cpython-36m-x86_64-linux-gnu.so(+0x11157)[0x7f5c88648157]
[task16227-q7qdf:09966] [13] /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/_raisim_gym.cpython-36m-x86_64-linux-gnu.so(+0x256be)[0x7f5c8865c6be]
[task16227-q7qdf:09966] [14] python(_PyCFunction_FastCallDict+0x35c)[0x56204c]
[task16227-q7qdf:09966] [15] python[0x4f88ba]
[task16227-q7qdf:09966] [16] python(_PyEval_EvalFrameDefault+0x467)[0x4f98c7]
[task16227-q7qdf:09966] [17] python(_PyFunction_FastCallDict+0xf5)[0x4f4065]
[task16227-q7qdf:09966] [18] python[0x5a1481]
[task16227-q7qdf:09966] [19] python[0x512a60]
[task16227-q7qdf:09966] [20] python[0x53ee21]
[task16227-q7qdf:09966] [21] python(_PyObject_FastCallKeywords+0x19c)[0x57ec0c]
[task16227-q7qdf:09966] [22] python[0x4f88ba]
[task16227-q7qdf:09966] [23] python(_PyEval_EvalFrameDefault+0x467)[0x4f98c7]
[task16227-q7qdf:09966] [24] python[0x4f6128]
[task16227-q7qdf:09966] [25] python(PyEval_EvalCode+0x23)[0x4f9023]
[task16227-q7qdf:09966] [26] python[0x6415b2]
[task16227-q7qdf:09966] [27] python(PyRun_FileExFlags+0x9a)[0x64166a]
[task16227-q7qdf:09966] [28] python(PyRun_SimpleFileExFlags+0x190)[0x643730]
[task16227-q7qdf:09966] [29] python(Py_Main+0x59e)[0x62b26e]
[task16227-q7qdf:09966] *** End of error message ***
[1]    9966 segmentation fault (core dumped)  python scripts/laikago_blind_locomotion.py
```
run "glxinfo | grep 'OpenGL render', output:
``` bash
OpenGL renderer string: llvmpipe (LLVM 7.0, 256 bits)
```

jhwangbo, 2019-08-19:
I think this is a well-known opengl/docker issue. You cannot run an OpenGL application inside a docker easily. Please check 'run_anymal_example.bash' file. If this doesn't solve your issue, let me know

jhwangbo, 2019-08-19:
Hi dytan, can you explain why you want to run the raisimGym with ssh? Are you using raisim in a remote machine? 

Frankly, I haven't used raisimGym in a remote machine. The only reason that it would not work is Ogre as far as I see. We will shortly start working on raisimUnity which will be a remote visualizer. Until then, I would recommend turning the rendering off in the config and make a separate video recorder (just like openai baselines).

dytan, 2019-08-19:
Thank you for the replay.

That's right, I try to run the env in remote machines. And the machine is on cluster, docker container dispatched by some process which I have no access to the 'nvidia-docker' command, everything is through ssh. Testing on local is OK.
Also tried [visualgl](https://www.virtualgl.org/), but havn't made it work yet.

Turing the rendering off shoud be fine.
Thank you again.

closed 2019-08-19 by jhwangbo

dytan, 2019-08-19:
Turn's out follow solution on the ogre [issue 810](https://github.com/OGRECave/ogre/issues/810) and rebuild ogre will make things work (tried this earlier on a ubuntu16.04 docker image (which opengl renderer string is 'software Rasterizer') encounter another exception).

alter ['ogre/RenderSystems/GLSupport/src/GLX/OgreGLXGLSupport.cpp: GLXGLSupport::GLXGLSupport()'])https://github.com/leggedrobotics/ogre/blob/a53f89c3870eb393631e325a8db25cef4f852820/RenderSystems/GLSupport/src/GLX/OgreGLXGLSupport.cpp#L93)
``` cpp
                mCurrentMode.width = screenSizes[currentSizeID].width;
                mCurrentMode.height = screenSizes[currentSizeID].height;
                mCurrentMode.refreshRate = XRRConfigCurrentRate(screenConfig);
```
to 
``` cpp

                if (nSizes==0){
                  mCurrentMode.width = DisplayWidth(mXDisplay, DefaultScreen(mXDisplay));
                  mCurrentMode.height = DisplayHeight(mXDisplay, DefaultScreen(mXDisplay));
                  mCurrentMode.refreshRate = 0;
                  mVideoModes.push_back(mCurrentMode);
                }
                else {
                  mCurrentMode.width = screenSizes[currentSizeID].width;
                  mCurrentMode.height = screenSizes[currentSizeID].height;
                  mCurrentMode.refreshRate = XRRConfigCurrentRate(screenConfig);
                }
```

[image]