leggedrobotics/raisimGym #6: Using Docker to run Tutorial - Configuration for Visualisation

haloted, 2019-08-20:
Hi Inkyu and Jemin,
I tried to use Docker container based on the original docker image created for the ANYmal and Lakaigo setup for this tutorial and managed to make it compile as it can run RL episodes with the following output:

> 
>     /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:516: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>     _np_qint8 = np.dtype([("qint8", np.int8, 1)])
>     /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:517: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>     _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
>     /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:518: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>     _np_qint16 = np.dtype([("qint16", np.int16, 1)])
>     /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:519: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>     _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
>     /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:520: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>     _np_qint32 = np.dtype([("qint32", np.int32, 1)])
>     /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:525: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>     np_resource = np.dtype([("resource", np.ubyte, 1)])
>     /usr/local/lib/python3.6/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:541: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>     _np_qint8 = np.dtype([("qint8", np.int8, 1)])
>     /usr/local/lib/python3.6/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:542: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>     _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
>     /usr/local/lib/python3.6/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:543: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>     _np_qint16 = np.dtype([("qint16", np.int16, 1)])
>     /usr/local/lib/python3.6/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:544: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>     _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
>     /usr/local/lib/python3.6/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:545: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>     _np_qint32 = np.dtype([("qint32", np.int32, 1)])
>     /usr/local/lib/python3.6/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:550: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>     np_resource = np.dtype([("resource", np.ubyte, 1)])
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
>     /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
[... cut ...]
I encountered two problems:
1. I was not able to visualise in Ogre as the raisimOgre window did not appear on the host computer.
2. The tensorboard also did not appear as there was no browser installed. So I installed firefox within the image. Initially it was able to trigger a firefox pop up in the host computer however it was not able to communicate and the webpage was broken. And now I am getting the error on cannot run firefox in root as a regular user problem.

Some guidance on this issue would be appreciated.

jhwangbo, 2019-08-20:
Hi Ted, this issue was posted a few days ago. I'll look into it. but I am not sure if this is even possible with Ogre so there is no guarantee. We will be working on raisimUnity soon, which will be a remote visualizer for this kind of problems. Until I find a solution, you can turn off the rendering in the config. You can still visually check the environment on your local machine.

haloted, 2019-08-20:
> Hi Ted, this issue was posted a few days ago. I'll look into it. but I am not sure if this is even possible with Ogre so there is no guarantee. We will be working on raisimUnity soon, which will be [...]

It kind of does not make sense that the docker configuration works for ANYmal and Laikago and not the Cartpole example. It seemed to be more of a configuration problem that I do not know about or it's more about the Docker network configuration that would be able to output the tensorboard on the host Firefox browser. 
Thanks for looking into this and let me know if there is a solution.

haloted, 2019-08-20:
UPDATE: The quadrotor example I have just tested worked well just like the ANYmal and Laikago example. However the second issue with trying to open Firefox for tensorboard remained.

jhwangbo, 2019-08-21:
Ted, i thought you are runnung it on a remote machine, arent you? If not, Ill check cart pole problem. It is probably missing a "render" option in the config file

haloted, 2019-08-21:
I am running it on docker just on my own laptop. I used docker because I may need to migrate the solution to the other laptops or machines running Optitrack.

haloted, 2019-08-21:
Is there a tutorial on how to playback training video and options related to reviewing training?

haloted, 2019-08-21:
> Ted, i thought you are runnung it on a remote machine, arent you? If not, Ill check cart pole problem. It is probably missing a "render" option in the config file
Yes adding the render:yes option solved the first problem. Cheers.

jhwangbo, 2019-08-21:
> Is there a tutorial on how to playback training video and options related to reviewing training?

I haven't tried this yet. I guess the current docker file doesn't even install ffmpeg. I'll try it out today

haloted, 2019-08-21:
> > Is there a tutorial on how to playback training video and options related to reviewing training?
> 
> I haven't tried this yet. I guess the current docker file doesn't even install ffmpeg. I'll try it out today

While you are at it, try installing Firefox and check if tensorboard can work in the docker environment. Thanks

haloted, 2019-08-21:
I also tried to run the last line of the cartpole tutorial after its training and found it was not able to load the environment for visualisation. The output of the error is provided as follows: 

> root@4462e3857ee9:/raisim_workspace/raisimGym# python3 ./cartpole_tutorial.py --mode test --weight ./weights/cartpole_tutorial/cartpole.pkl
> python3: can't open file './cartpole_tutorial.py': [Errno 2] No such file or directory
> root@4462e3857ee9:/raisim_workspace/raisimGym# cd raisim_gym/env/env/cartpole/
> root@4462e3857ee9:/raisim_workspace/raisimGym/raisim_gym/env/env/cartpole# python3 ./cartpole_tutorial.py --mode test --weight ./weights/cartpole_tutorial/cartpole.pkl
> /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:516: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>   _np_qint8 = np.dtype([("qint8", np.int8, 1)])
> /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:517: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>   _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
> /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:518: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>   _np_qint16 = np.dtype([("qint16", np.int16, 1)])
> /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:519: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>   _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
> /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:520: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>   _np_qint32 = np.dtype([("qint32", np.int32, 1)])
> /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:525: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>   np_resource = np.dtype([("resource", np.ubyte, 1)])
> /usr/local/lib/python3.6/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:541: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>   _np_qint8 = np.dtype([("qint8", np.int8, 1)])
> /usr/local/lib/python3.6/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:542: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>   _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
> /usr/local/lib/python3.6/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:543: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>   _np_qint16 = np.dtype([("qint16", np.int16, 1)])
> /usr/local/lib/python3.6/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:544: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>   _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
> /usr/local/lib/python3.6/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:545: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>   _np_qint32 = np.dtype([("qint32", np.int32, 1)])
> /usr/local/lib/python3.6/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:550: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
>   np_resource = np.dtype([("resource", np.ubyte, 1)])
> /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
> [2019:08:21:08:35:15 OgreVis.hpp:295] Loading RaisimOgre Resource from: /raisim_build/share/raisimOgre/cmake/../rsc
> [2019:08:21:08:35:15 OgreVis.hpp:296] Loading Ogre Configuration from: /raisim_build/share/raisimOgre/cmake/../ogre/
> Loaded weight from ./weights/cartpole_tutorial/cartpole.pkl
> 
> Loading a model without an environment, this model cannot be trained until it has a valid environment.
> WARNING: Logging before flag parsing goes to stderr.
> W0821 08:35:16.083068 140351034369856 deprecation_wrapper.py:119] From /usr/local/lib/python3.6/dist-packages/stable_baselines-2.7.0-py3.6.egg/stable_baselines/common/tf_util.py:98: The name tf.ConfigProto is deprecated. Please use tf.compat.v1.ConfigProto instead.
> 
> W0821 08:35:16.083539 140351034369856 deprecation_wrapper.py:119] From /usr/local/lib/python3.6/dist-packages/stable_baselines-2.7.0-py3.6.egg/stable_baselines/common/tf_util.py:107: The name tf.Session is deprecated. Please use tf.compat.v1.Session instead.
> 
[... cut ...]
>   File "./cartpole_tutorial.py", line 79, in <module>
>     obs, reward, done, infos = env.step(action, visualize=True)
>   File "/usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/RaisimGymVecEnv.py", line 30, in step
>     self.wrapper.testStep(action, self._observation, self._reward, self._done, self._extraInfo)
> KeyboardInterrupt

The error is as it shows cannot load a valid environment and so the RaisimOgre window only popped open briefly. Please also look into this issue as well. Cheers.

jhwangbo, 2019-08-21:
now the docker file installs firefox and displays tensorboard on it. Please keep different issues separate. This new issue should be post separately

haloted, 2019-08-22:
Hi Jemin, I tried running the ANYmal example runner.bash file, I do get firefox to pop up but for some reason it still did not display the tensorboard. The output is :

> Hmm. We’re having trouble finding that site.
> 
> > We can’t connect to the server at b09379db7dd3.
> > 
> > If that address is correct, here are three other things you can try:
> > 
> >     Try again later.
> >     Check your network connection.
> >     If you are connected but behind a firewall, check that Firefox has permission to access the Web.

[image]

jhwangbo, 2019-08-22:
Hi Ted, I can't figure out this issue because it works fine on my PC. Can you fix this issue and send a pull request?

haloted, 2019-08-22:
Hi Jemin, did you use google chrome for your browser inside docker? 
I found that google chrome cannot be used as it returns the --no-sandbox unsupported errors. Was your tensorboard run on firefox browser?

jhwangbo, 2019-08-22:
Hi Ted, I used the run_anymal_example.sh as is. I didn't modify it. it
works in firefox

Jemin Hwangbo
Swiss Federal Institute of Technology (ETHZ)
Robotic Systems Lab (RSL)
LEE H315, Leonhardstr. 21
CH-8092 Zurich, Switzerland
http://rsl.ethz.ch/
<https://mail.ethz.ch/owa/redir.aspx?C=Gg_LcRNKWkWJtDW86tGsCG5c-cz8l9EInOTRkp4V-DV9W0bgUJnG41ysxMv262id7Ag2dL8n6y4.&URL=http%3a%2f%2fleggedrobotics.ethz.ch%2f>

Office: +41 44 632 99 78
Fax:    +41 44 632 11 81

jhwangbo, 2019-08-22:
Hi Ted, i used the example as is. I works with firefix

jhwangbo, 2019-08-24:
can't reproduce the issue. closing the issue for now

closed 2019-08-24 by jhwangbo

rmbp, 2019-09-05:
> I also tried to run the last line of the cartpole tutorial after its training and found it was not able to load the environment for visualisation. The output of the error is provided as follows:
> 
> > root@4462e3857ee9:/raisim_workspace/raisimGym# python3 ./cartpole_tutorial.py --mode test --weight ./weights/cartpole_tutorial/cartpole.pkl
> > python3: can't open file './cartpole_tutorial.py': [Errno 2] No such file or directory
> > root@4462e3857ee9:/raisim_workspace/raisimGym# cd raisim_gym/env/env/cartpole/
> > root@4462e3857ee9:/raisim_workspace/raisimGym/raisim_gym/env/env/cartpole# python3 ./cartpole_tutorial.py --mode test --weight ./weights/cartpole_tutorial/cartpole.pkl
> > /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:516: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
> > _np_qint8 = np.dtype([("qint8", np.int8, 1)])
> > /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:517: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
> > _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
> > /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:518: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
> > _np_qint16 = np.dtype([("qint16", np.int16, 1)])
> > /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:519: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
> > _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
> > /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:520: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
> > _np_qint32 = np.dtype([("qint32", np.int32, 1)])
> > /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:525: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
> > np_resource = np.dtype([("resource", np.ubyte, 1)])
> > /usr/local/lib/python3.6/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:541: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
> > _np_qint8 = np.dtype([("qint8", np.int8, 1)])
> > /usr/local/lib/python3.6/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:542: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
> > _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
> > /usr/local/lib/python3.6/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:543: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
> > _np_qint16 = np.dtype([("qint16", np.int16, 1)])
> > /usr/local/lib/python3.6/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:544: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
> > _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
> > /usr/local/lib/python3.6/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:545: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
> > _np_qint32 = np.dtype([("qint32", np.int32, 1)])
> > /usr/local/lib/python3.6/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:550: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
> > np_resource = np.dtype([("resource", np.ubyte, 1)])
> > /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
> > [2019:08:21:08:35:15 OgreVis.hpp:295] Loading RaisimOgre Resource from: /raisim_build/share/raisimOgre/cmake/../rsc
> > [2019:08:21:08:35:15 OgreVis.hpp:296] Loading Ogre Configuration from: /raisim_build/share/raisimOgre/cmake/../ogre/
> > Loaded weight from ./weights/cartpole_tutorial/cartpole.pkl
> > Loading a model without an environment, this model cannot be trained until it has a valid environment.
> > WARNING: Logging before flag parsing goes to stderr.
> > W0821 08:35:16.083068 140351034369856 deprecation_wrapper.py:119] From /usr/local/lib/python3.6/dist-packages/stable_baselines-2.7.0-py3.6.egg/stable_baselines/common/tf_util.py:98: The name tf.ConfigProto is deprecated. Please use tf.compat.v1.ConfigProto instead.
> > W0821 08:35:16.083539 140351034369856 deprecation_wrapper.py:119] From /usr/local/lib/python3.6/dist-packages/stable_baselines-2.7.0-py3.6.egg/stable_baselines/common/tf_util.py:107: The name tf.Session is deprecated. Please use tf.compat.v1.Session instead.
[... cut ...]
> > File "/usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/RaisimGymVecEnv.py", line 30, in step
> > self.wrapper.testStep(action, self._observation, self._reward, self._done, self._extraInfo)
> > KeyboardInterrupt
> 
> The error is as it shows cannot load a valid environment and so the RaisimOgre window only popped open briefly. Please also look into this issue as well. Cheers.

Hi Ted,

I'm also having the same issue on my own custom env I've built.  Have you had any luck with displaying test cases as per the last line of the cartpole tutorial? This is on my own laptop.

rmbp, 2019-09-05:
Hi both,

I also have the issue of test cases not displaying as per the last line of the cartpole tutorial.  The orge window pops up then closes before any visualisation loads, but the model runs fine, outputting episode rewards etc.

haloted, 2019-09-05:
Hi rmbp, have you tried looking into the cartpole default configuration file.yaml in the rsc folder. 
See if it is missing a render option. If it is missing one, add it as render: yes. Then it should be able to visualise.

rmbp, 2019-09-05:
It is in there, and if i change it to render: no, the window does not pop up at all.

jhwangbo, 2019-09-05:
The screen initially pops up and closes. but then it should pop up again after ```env.wrapper.showWindow()```. I just pulled it fresh and tested it on my machine. Are you running it on a remote device? Remote device rendering is not supported now. We are working on it right now.