leggedrobotics/raisimGym #7: Cartpole Tutorial Testing Mode

haloted, 2019-08-22:
Also tried to run the last line of the cartpole tutorial after its training and found it was not able to load the environment for visualisation. The output of the error is provided as follows:

    root@4462e3857ee9:/raisim_workspace/raisimGym# python3 ./cartpole_tutorial.py --mode test --weight ./weights/cartpole_tutorial/cartpole.pkl
    python3: can't open file './cartpole_tutorial.py': [Errno 2] No such file or directory
    root@4462e3857ee9:/raisim_workspace/raisimGym# cd raisim_gym/env/env/cartpole/
    root@4462e3857ee9:/raisim_workspace/raisimGym/raisim_gym/env/env/cartpole# python3 ./cartpole_tutorial.py --mode test --weight ./weights/cartpole_tutorial/cartpole.pkl
    /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:516: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
    _np_qint8 = np.dtype([("qint8", np.int8, 1)])
    /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:517: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
    _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
    /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:518: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
    _np_qint16 = np.dtype([("qint16", np.int16, 1)])
    /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:519: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
    _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
    /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:520: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
    _np_qint32 = np.dtype([("qint32", np.int32, 1)])
    /usr/local/lib/python3.6/dist-packages/tensorflow/python/framework/dtypes.py:525: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
    np_resource = np.dtype([("resource", np.ubyte, 1)])
    /usr/local/lib/python3.6/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:541: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
    _np_qint8 = np.dtype([("qint8", np.int8, 1)])
    /usr/local/lib/python3.6/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:542: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
    _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
    /usr/local/lib/python3.6/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:543: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
    _np_qint16 = np.dtype([("qint16", np.int16, 1)])
    /usr/local/lib/python3.6/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:544: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
    _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
    /usr/local/lib/python3.6/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:545: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
    _np_qint32 = np.dtype([("qint32", np.int32, 1)])
    /usr/local/lib/python3.6/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:550: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
    np_resource = np.dtype([("resource", np.ubyte, 1)])
    /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/cartpole/rsc
    [2019:08:21:08:35:15 OgreVis.hpp:295] Loading RaisimOgre Resource from: /raisim_build/share/raisimOgre/cmake/../rsc
    [2019:08:21:08:35:15 OgreVis.hpp:296] Loading Ogre Configuration from: /raisim_build/share/raisimOgre/cmake/../ogre/
    Loaded weight from ./weights/cartpole_tutorial/cartpole.pkl

    Loading a model without an environment, this model cannot be trained until it has a valid environment.
    WARNING: Logging before flag parsing goes to stderr.
    W0821 08:35:16.083068 140351034369856 deprecation_wrapper.py:119] From /usr/local/lib/python3.6/dist-packages/stable_baselines-2.7.0-py3.6.egg/stable_baselines/common/tf_util.py:98: The name tf.ConfigProto is deprecated. Please use tf.compat.v1.ConfigProto instead.

    W0821 08:35:16.083539 140351034369856 deprecation_wrapper.py:119] From /usr/local/lib/python3.6/dist-packages/stable_baselines-2.7.0-py3.6.egg/stable_baselines/common/tf_util.py:107: The name tf.Session is deprecated. Please use tf.compat.v1.Session instead.

    W0821 08:35:16.189958 140351034369856 deprecation_wrapper.py:119] From /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/archi/policies.py:112: The name tf.variable_scope is deprecated. Please use tf.compat.v1.variable_scope instead.

    W0821 08:35:16.190448 140351034369856 deprecation_wrapper.py:119] From /usr/local/lib/python3.6/dist-packages/stable_baselines-2.7.0-py3.6.egg/stable_baselines/common/input.py:25: The name tf.placeholder is deprecated. Please use tf.compat.v1.placeholder instead.

    W0821 08:35:16.191884 140351034369856 deprecation.py:323] From /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/archi/policies.py:436: flatten (from tensorflow.python.layers.core) is deprecated and will be removed in a future version.
    Instructions for updating:
    Use keras.layers.flatten instead.
    W0821 08:35:16.454259 140351034369856 deprecation_wrapper.py:119] From /usr/local/lib/python3.6/dist-packages/stable_baselines-2.7.0-py3.6.egg/stable_baselines/a2c/utils.py:156: The name tf.get_variable is deprecated. Please use tf.compat.v1.get_variable instead.

    W0821 08:35:16.666865 140351034369856 deprecation_wrapper.py:119] From /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/algo/ppo2/ppo2.py:161: The name tf.summary.scalar is deprecated. Please use tf.compat.v1.summary.scalar instead.

    W0821 08:35:16.693435 140351034369856 deprecation.py:323] From /usr/local/lib/python3.6/dist-packages/tensorflow/python/ops/math_grad.py:1250: add_dispatch_support..wrapper (from tensorflow.python.ops.array_ops) is deprecated and will be removed in a future version.
    Instructions for updating:
    Use tf.where in 2.0, which has the same broadcast rule as np.where
    W0821 08:35:16.920613 140351034369856 deprecation_wrapper.py:119] From /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/algo/ppo2/ppo2.py:177: The name tf.train.AdamOptimizer is deprecated. Please use tf.compat.v1.train.AdamOptimizer instead.

    W0821 08:35:17.114641 140351034369856 deprecation_wrapper.py:119] From /usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/algo/ppo2/ppo2.py:210: The name tf.summary.merge_all is deprecated. Please use tf.compat.v1.summary.merge_all instead.

    ^CTraceback (most recent call last):
    File "./cartpole_tutorial.py", line 79, in
    obs, reward, done, infos = env.step(action, visualize=True)
    File "/usr/local/lib/python3.6/dist-packages/raisim_gym-0.3.0-py3.6-linux-x86_64.egg/raisim_gym/env/RaisimGymVecEnv.py", line 30, in step
    self.wrapper.testStep(action, self._observation, self._reward, self._done, self._extraInfo)
    KeyboardInterrupt

The error is as it shows cannot load a valid environment and so the RaisimOgre window only popped open briefly. Please also look into this issue as well. Cheers.

jhwangbo, 2019-08-22:
Hi Ted, this error was due to not opening visialization window manially. Now recroding video and opening an window should be done manually. Please pull and try again. It is fixed now

haloted, 2019-08-22:
How do I open the visualisation window manually? Do I have to run a command to open raisimOgre? As far as I know, the visualisation window is opened automatically during simulation and closes after an episode is run.

jhwangbo, 2019-08-22:
pull raisimGymTutorial and see what has changed. The change is already made there. You can run the example now with the visualization

closed 2019-08-22 by jhwangbo

haloted, 2019-08-22:
Thanks, it now works.

haloted, 2019-09-14:
For the cartpole tutorial during testing, there were additional method calls employed by the PPO2 class, one is model.load(weights) (PPO2) and one is model.predict().

However it seems that it was not included in the official raisimGym Repo. Would you know where I can find their definitions?

haloted, 2019-09-17:
I found the model.load(weights path) and model.predict() are defined under the class ActorCriticRLModel(BaseRLModel) under the stable-baseline package where the PPO2 class is inherited from.