leggedrobotics/raisimGym #8: Trying to Build Environment Error

haloted, 2019-08-29:
Hi Jemin,

Out of the blue this error occurred when I try to rebuild environment using the code: 
`python3 setup.py install --CMAKE_PREFIX_PATH $LOCAL_BUILD --env ANYmal
`or 
`python3 setup.py install --CMAKE_PREFIX_PATH $LOCAL_BUILD --env cartpole
`
Here is the output: 

```
> `running install
> running bdist_egg
> running egg_info
> writing raisim_gym.egg-info/PKG-INFO
> writing dependency_links to raisim_gym.egg-info/dependency_links.txt
> writing requirements to raisim_gym.egg-info/requires.txt
> writing top-level names to raisim_gym.egg-info/top_level.txt
> reading manifest file 'raisim_gym.egg-info/SOURCES.txt'
> reading manifest template 'MANIFEST.in'
> writing manifest file 'raisim_gym.egg-info/SOURCES.txt'
> installing library code to build/bdist.linux-x86_64/egg
> running install_lib
> running build_py
> running build_ext
> CMake Error at CMakeLists.txt:7 (find_package):
>   Could not find a configuration file for package "raisimOgre" that is
>   compatible with requested version "0.4.0".
> 
>   The following configuration files were considered but not accepted:
> 
>     /raisim_build/share/raisimOgre/cmake/raisimOgreConfig.cmake, version: 0.3.0
> 
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/raisim_workspace/raisimGym/build/temp.linux-x86_64-3.6/CMakeFiles/CMakeOutput.log".
> Traceback (most recent call last):
>   File "setup.py", line 104, in <module>
>     zip_safe=False,
>   File "/usr/lib/python3.6/distutils/core.py", line 148, in setup
>     dist.run_commands()
>   File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
>     self.run_command(cmd)
>   File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
>     cmd_obj.run()
>   File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 67, in run
>     self.do_egg_install()
>   File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 109, in do_egg_install
>     self.run_command('bdist_egg')
>   File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
>     self.distribution.run_command(command)
>   File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
>     cmd_obj.run()
>   File "/usr/lib/python3/dist-packages/setuptools/command/bdist_egg.py", line 172, in run
>     cmd = self.call_command('install_lib', warn_dir=0)
>   File "/usr/lib/python3/dist-packages/setuptools/command/bdist_egg.py", line 158, in call_command
>     self.run_command(cmdname)
>   File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
>     self.distribution.run_command(command)
>   File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
>     cmd_obj.run()
>   File "/usr/lib/python3/dist-packages/setuptools/command/install_lib.py", line 24, in run
>     self.build()
>   File "/usr/lib/python3.6/distutils/command/install_lib.py", line 109, in build
>     self.run_command('build_ext')
>   File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
>     self.distribution.run_command(command)
>   File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
>     cmd_obj.run()
>   File "setup.py", line 60, in run
>     self.build_extension(ext)
>   File "setup.py", line 88, in build_extension
>     subprocess.check_call(['cmake', ext.sourcedir] + cmake_args, cwd=self.build_temp, env=env)
>   File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
>     raise CalledProcessError(retcode, cmd)
> subprocess.CalledProcessError: Command '['cmake', '/raisim_workspace/raisimGym', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/raisim_workspace/raisimGym/build/lib.linux-x86_64-3.6', '-DPYTHON_EXECUTABLE=/usr/bin/python3', '-DCMAKE_PREFIX_PATH=/raisim_build', '-DRSG_ENVIRONMENT_INCLUDE_PATH=ANYmal', '-DCMAKE_BUILD_TYPE=Release']' returned non-zero exit status 1.
> `
```

Please help.

jhwangbo, 2019-08-29:
The error is this line

```
>   Could not find a configuration file for package "raisimOgre" that is
>   compatible with requested version "0.4.0".
```

Simply means that you don't have the right version. Can you pull raisimLib and raisimOgre, install them again and compile raisimGym?

haloted, 2019-08-29:
I used git pull in both of my folders raisimOgre and raisimLib and they are all up to date. But I will try the whole reinstallation process to see if it changes anything.

jhwangbo, 2019-08-29:
I often find that cmake doesn't update config files even when it should update. An easy way to fix it is to go to $LOCAL_BUILD/share and delete the raisimOgre folder manually. Then reinstall

haloted, 2019-08-29:
I deleted the raisimOgre and now trying to install raisimOgre by cloning from git into the workspace directory. When I try to run the installation within the build folder: 
```
cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_PREFIX_PATH=$LOCAL_BUILD -DCMAKE_INSTALL_PREFIX=$LOCAL_BUILD
make install -j8
I got the following error output: 

```
> CMake Error at CMakeLists.txt:12 (find_package):
>   Could not find a package configuration file provided by "raisim" (requested
>   version 0.4.0) with any of the following names:
> 
>     raisimConfig.cmake
>     raisim-config.cmake
> 
>   Add the installation prefix of "raisim" to CMAKE_PREFIX_PATH or set
>   "raisim_DIR" to a directory containing one of the above files.  If "raisim"
>   provides a separate development package or SDK, be sure it has been
>   installed.
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/raisim_workspace/raisimOgre/build/CMakeFiles/CMakeOutput.log".
```

```

haloted, 2019-08-29:
I actually did try removing the raisim_build folder completely and go through the installation for Ogre first and then raisimOgre until I got this problem ^

jhwangbo, 2019-08-29:
how about raisim then? did you reinstall raisim after you delete raisim_build folder?

jhwangbo, 2019-08-29:
This issue seems not related to raisim. closing it until I get a further response

closed 2019-08-29 by jhwangbo

haloted, 2019-08-30:
I tried reinstall everything from the beginning deleting both the raisim_workspace and raisim_build folders. Now I was able to compile the anymal example but for some reason it cannot detect the Environmental.hpp for both lakaigo and hummingbird when I tried to build environment using this line: 
`python3 setup.py install --CMAKE_PREFIX_PATH $LOCAL_BUILD --env hummingbird`

The output: 

> running install
> running bdist_egg
> running egg_info
> writing raisim_gym.egg-info/PKG-INFO
> writing dependency_links to raisim_gym.egg-info/dependency_links.txt
> writing requirements to raisim_gym.egg-info/requires.txt
> writing top-level names to raisim_gym.egg-info/top_level.txt
> reading manifest file 'raisim_gym.egg-info/SOURCES.txt'
> reading manifest template 'MANIFEST.in'
> writing manifest file 'raisim_gym.egg-info/SOURCES.txt'
> installing library code to build/bdist.linux-x86_64/egg
> running install_lib
> running build_py
> running build_ext
> -- raisim:
> --   Version: 0.4.2
> --   Includes: /raisim_build/include
> --   Libraries: raisim::png;raisim::raisimODE
> -- Found OGRE
> --   static     : OFF
> --   components : Bites;HLMS;MeshLodGenerator;Overlay;Paging;Property;RTShaderSystem;Terrain;Volume
> --   plugins    : Plugin_BSPSceneManager;Plugin_OctreeSceneManager;Plugin_PCZSceneManager;Plugin_ParticleFX;RenderSystem_GL;RenderSystem_GL3Plus;Codec_STBI;Codec_FreeImage
> --   media      : /raisim_build/share/OGRE/Media
> -- raisimOgre:
> --   Version: 0.4.0
> --   Includes: /raisim_build/include
> --   Libraries: OgreMain;OgreBites;OgreRTShaderSystem;OgreMeshLodGenerator;raisim::assimp;raisim::raisim;pthread;zzip
> [RAISIM_GYM] including directory hummingbird
> -- Configuring done
> -- Generating done
> -- Build files have been written to: /raisim_workspace/raisimGym/build/temp.linux-x86_64-3.6
> [ 50%] Building CXX object CMakeFiles/_raisim_gym.dir/raisim_gym/env/raisim_gym.cpp.o
> /raisim_workspace/raisimGym/raisim_gym/env/raisim_gym.cpp:4:10: fatal error: Environment.hpp: No such file or directory
>  #include "Environment.hpp"
>           ^~~~~~~~~~~~~~~~~
> compilation terminated.
> CMakeFiles/_raisim_gym.dir/build.make:62: recipe for target 'CMakeFiles/_raisim_gym.dir/raisim_gym/env/raisim_gym.cpp.o' failed
> make[2]: *** [CMakeFiles/_raisim_gym.dir/raisim_gym/env/raisim_gym.cpp.o] Error 1
> CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/_raisim_gym.dir/all' failed
> make[1]: *** [CMakeFiles/_raisim_gym.dir/all] Error 2
> Makefile:83: recipe for target 'all' failed
> make: *** [all] Error 2
> Traceback (most recent call last):
>   File "setup.py", line 101, in <module>
>     zip_safe=False,
>   File "/usr/lib/python3.6/distutils/core.py", line 148, in setup
>     dist.run_commands()
>   File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
>     self.run_command(cmd)
>   File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
>     cmd_obj.run()
>   File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 67, in run
>     self.do_egg_install()
>   File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 109, in do_egg_install
>     self.run_command('bdist_egg')
>   File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
>     self.distribution.run_command(command)
>   File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
>     cmd_obj.run()
>   File "/usr/lib/python3/dist-packages/setuptools/command/bdist_egg.py", line 172, in run
>     cmd = self.call_command('install_lib', warn_dir=0)
>   File "/usr/lib/python3/dist-packages/setuptools/command/bdist_egg.py", line 158, in call_command
>     self.run_command(cmdname)
>   File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
>     self.distribution.run_command(command)
>   File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
>     cmd_obj.run()
>   File "/usr/lib/python3/dist-packages/setuptools/command/install_lib.py", line 24, in run
>     self.build()
>   File "/usr/lib/python3.6/distutils/command/install_lib.py", line 109, in build
>     self.run_command('build_ext')
>   File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
>     self.distribution.run_command(command)
>   File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
>     cmd_obj.run()
>   File "setup.py", line 57, in run
>     self.build_extension(ext)
>   File "setup.py", line 86, in build_extension
>     subprocess.check_call(['cmake', '--build', '.'] + build_args, cwd=self.build_temp)
>   File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
>     raise CalledProcessError(retcode, cmd)
> subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release', '--', '-j2']' returned non-zero exit status 2.
>

haloted, 2019-08-30:
The above problem is quite strange, I tried adding a / at the end of the folder directory for lakaigo and hummingbird and it worked. However for ANYmal a / wasn't required for the directory path for it to compile.

haloted, 2019-08-30:
I managed to get it to compile after reinstalling from scratch by deleting the raisim_workspace folder and the raisim_build folders.