leggedrobotics/raisimGym #10: Deploying Trained Weights for Quadcopter

haloted, 2019-08-30:
Hi Jemin,

I was wondering in the quadcopter simulation, we can set desired simulation time and iterations. I was wondering whether we have to wait for the simulation to finish completely before the trained weights information is saved in the data folder? 
If we want to deploy the weights, can we just use the same code in the Cartpole_Tutorial.py? 
From my understanding of the code, there doesn't seemed to be any line that points to saving the configuration apart from the saver.data_dir to save the configuration from tensorboard. But what if I wanted to make a weights folder to save the quadcopter trained weights?

jhwangbo, 2019-08-30:
I think the cleanest way to save intermediate params is to use the callbacks. You can also simply input a line in the PPO2. raisimGym is not meant to be a learning framework. It is a collection of examples of wrapping raisim in python.

I don't get the second part of your question. If can make your own folder for saving NN params.

haloted, 2019-08-30:
I guess what I really mean is how to obtain the .pkl file which contains the weights from training. From my understanding, the dump( ) method is the way to go. What I am unsure about is whether the .pkl file is created during the simulation or after the simulation, ie. I can only find the .pkl file after the set simulation time is reached. 
And if I need to save the NN parameters during simulation, saving the .pkl in real time, what is the code implementation?

jhwangbo, 2019-08-30:
The example is in the tutorial. You can do it with a callback as I said on top.

closed 2019-08-30 by jhwangbo

haloted, 2019-09-02:
I am trying to find where the PPO2.load method is, it doesn't seem to be included in the PPO2.py algorithm file. However it is used in the cartpole_tutorial.py file and it works for simulating the weights based on the weights policy? 

What I am trying to do is to load the existing trained weights and continue the model.learn() process to continue optimising the weights and save it after a set iteration. Is this possible?