leggedrobotics/raisimGym #9: Firefox installed in docker cannot be run as sudo based on UID problem

haloted, 2019-08-30:
It is probably an old issue but firefox when initially installed using the dockerfile provided runs okay. Sometimes I still get the error that I cannot run firefox due to the wrong UID error as shown below, as this also affects the tensorboard operation: 
```

root@d22c0ccad313:/raisim_workspace/raisimGym# firefox
Running Firefox as root in a regular user's session is not supported.  ($XAUTHORITY is /tmp/.docker.xauth which is owned by uid 1000.)
```
A solution to this would be greatly appreciated.

jhwangbo, 2019-08-30:
As I said last time, feel free to contribute. I cannot reproduce this error

closed 2019-08-30 by jhwangbo

haloted, 2019-09-01:
Hi Jemin,
I downloaded another browser called iceweasel and it gave me the tensorboard screen. However it needs to read data from the data folder as shown below. Does this mean the website is still not fully connected? Or what do I need to do to make this work? 

[image]

jhwangbo, 2019-09-01:
Ted, just open the browser on your host PC. This is an overkill and a waste of time.

haloted, 2019-09-06:
I have migrated everything onto my host PC and Initially I was able to run tensorboard no problems. 
After I installed ROS and I am not sure why or what I have changed but somehow I have to add sudo to run the build environment command as well as the sudo run example script. And also owing to this system change, the tensorboard now stops updating information on the web browser. 
It is saying that because python installation is in the root folder /usr/local/etc/python
I am now getting the error:  

> [RAISIM_GYM] Tensorboard session created: http://ted-Predator-G3-572:6006/
> WARNING:tensorflow:From /usr/local/lib/python3.5/dist-packages/stable_baselines-2.7.0-py3.5.egg/stable_baselines/common/base_class.py:880: The name tf.summary.FileWriter is deprecated. Please use tf.compat.v1.summary.FileWriter instead.
> 
> [8789:8789:0906/213216.760637:ERROR:zygote_host_impl_linux.cc(89)] Running as root without --no-sandbox is not supported. See https://crbug.com/638180.
> [RAISIM_GYM] Visualizing in RaiSimOgre
> [8799:8799:0906/213216.789773:ERROR:zygote_host_impl_linux.cc(89)] Running as root without --no-sandbox is not supported. See https://crbug.com/638180.
> Running Firefox as root in a regular user's session is not supported.  ($HOME is /home/ted which is owned by ted.)
> ffmpeg version 4.1.3-0york1~16.04 Copyright (c) 2000-2019 the FFmpeg developers
>   built with gcc 5.4.0 (Ubuntu 5.4.0-6ubuntu1~16.04.11) 20160609
>   configuration: --prefix=/usr --extra-version='0york1~16.04' --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-nonfree --enable-libfdk-aac --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
>   libavutil      56. 22.100 / 56. 22.100
>   libavcodec     58. 35.100 / 58. 35.100
>   libavformat    58. 20.100 / 58. 20.100
>   libavdevice    58.  5.100 / 58.  5.100
>   libavfilter     7. 40.101 /  7. 40.101
>   libavresample   4.  0.  0 /  4.  0.  0
>   libswscale      5.  3.100 /  5.  3.100
>   libswresample   3.  3.100 /  3.  3.100
>   libpostproc    55.  3.100 / 55.  3.100
> Input #0, rawvideo, from 'pipe:':
>   Duration: N/A, start: 0.000000, bitrate: 1327104 kb/s
>     Stream #0:0: Video: rawvideo (RGB[24] / 0x18424752), rgb24, 1280x720, 1327104 kb/s, 60 tbr, 60 tbn, 60 tbc
> Stream mapping:
>   Stream #0:0 -> #0:0 (rawvideo (native) -> h264 (libx264))
> [libx264 @ 0x5568dbb2ddc0] using cpu capabilities: MMX2 SSE2Fast SSSE3 SSE4.2 AVX FMA3 BMI2 AVX2
> [libx264 @ 0x5568dbb2ddc0] profile High, level 3.2
> [libx264 @ 0x5568dbb2ddc0] 264 - core 155 r2917 0a84d98 - H.264/MPEG-4 AVC codec - Copyleft 2003-2018 - http://www.videolan.org/x264.html - options: cabac=1 ref=2 deblock=1:0:0 analyse=0x3:0x113 me=hex subme=6 psy=1 psy_rd=1.00:0.00 mixed_ref=1 me_range=16 chroma_me=1 trellis=1 8x8dct=1 cqm=0 deadzone=21,11 fast_pskip=1 chroma_qp_offset=-2 threads=12 lookahead_threads=2 sliced_threads=0 nr=0 decimate=1 interlaced=0 bluray_compat=0 constrained_intra=0 bframes=3 b_pyramid=2 b_adapt=1 b_bias=0 direct=1 weightb=1 open_gop=0 weightp=1 keyint=250 keyint_min=25 scenecut=40 intra_refresh=0 rc_lookahead=30 rc=crf mbtree=1 crf=21.0 qcomp=0.60 qpmin=0 qpmax=69 qpstep=4 ip_ratio=1.40 aq=1:1.00
> Output #0, mp4, to '/home/ted/raisim_workspace/raisimGym/scripts/..//QuadrotorTrainingdata/quadrotor_position_tracking/2019-09-06-21-32-14/0.mp4':
>   Metadata:
>     encoder         : Lavf58.20.100
>     Stream #0:0: Video: h264 (libx264) (avc1 / 0x31637661), yuv420p, 1280x720, q=-1--1, 60 fps, 15360 tbn, 60 tbc
>     Metadata:
>       encoder         : Lavc58.35.100 libx264
>     Side data:
>       cpb: bitrate max/min/avg: 0/0/0 buffer size: 0 vbv_delay: -1
> frame=  334 fps= 60 q=-1.0 Lsize=     591kB time=00:00:05.51 bitrate= 878.1kbits/s speed=0.993x    
> video:587kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.776016%
> [libx264 @ 0x5568dbb2ddc0] frame I:3     Avg QP:16.75  size: 23388
> [libx264 @ 0x5568dbb2ddc0] frame P:107   Avg QP:24.47  size:  3342
> [libx264 @ 0x5568dbb2ddc0] frame B:224   Avg QP:25.19  size:   770
> [libx264 @ 0x5568dbb2ddc0] consecutive B-frames:  8.7%  4.2%  4.5% 82.6%
> [libx264 @ 0x5568dbb2ddc0] mb I  I16..4: 60.2% 23.0% 16.8%
> [libx264 @ 0x5568dbb2ddc0] mb P  I16..4:  1.1%  4.9%  1.7%  P16..4:  5.2%  3.5%  1.0%  0.0%  0.0%    skip:82.6%
> [libx264 @ 0x5568dbb2ddc0] mb B  I16..4:  0.5%  0.2%  0.1%  B16..8:  2.4%  1.5%  0.1%  direct: 0.2%  skip:95.1%  L0:51.4% L1:44.7% BI: 3.9%
[... cut ...]
> | ep_reward_mean     | -5.39          |
> | explained_variance | 0.928          |
> | fps                | 123907         |
> | nupdates           | 2              |
> | policy_entropy     | 5.665099       |
> | policy_loss        | -0.00046076067 |
> | serial_timesteps   | 1000           |
> | time_elapsed       | 8.29           |
> | total_timesteps    | 500020         |
> | value_loss         | 0.017508285    |
> ---------------------------------------

Any idea how to revert to the original settings to get tensorboard to work?

haloted, 2019-09-06:
Here is an error output from just using the normal command:

> python setup.py install --CMAKE_PREFIX_PATH $LOCAL_BUILD --env quadrotor
> running install
> error: can't create or remove files in install directory
> 
> The following error occurred while trying to add or remove files in the
> installation directory:
> 
>     [Errno 13] Permission denied: '/usr/local/lib/python2.7/dist-packages/test-easy-install-10127.write-test'
> 
> The installation directory you specified (via --install-dir, --prefix, or
> the distutils default setting) was:
> 
>     /usr/local/lib/python2.7/dist-packages/
> 
> Perhaps your account does not have write access to this directory?  If the
> installation directory is a system-owned directory, you may need to sign in
> as the administrator or "root" account.  If you do not have administrative
> access to this machine, you may wish to choose a different installation
> directory, preferably one that is listed in your PYTHONPATH environment
> variable.
> 
> For information on other options, you may wish to consult the
> documentation at:
> 
>   https://pythonhosted.org/setuptools/easy_install.html
> 
> Please make the appropriate changes for your system and try again.
>

jhwangbo, 2019-09-06:
Hi Ted, are you using a virtualenv?

haloted, 2019-09-06:
I don't think it is a virtual environment so I'm just using a normal python interpreter from /usr/bin/python
Am I supposed to use a virtual environment interpreter?

jhwangbo, 2019-09-06:
Yes. virtualenv or anaconda. I only use virtualenv but my colleagues also use anaconda and it works with raisimgym.

haloted, 2019-09-06:
I've used the virtual environment to run it and it still says permission denied... :(

> (venv) ted@ted-Predator-G3-572:~/raisim_workspace/raisimGym$ python3 setup.py install --CMAKE_PREFIX_PATH $LOCAL_BUILD --env quadrotor
> running install
> running bdist_egg
> running egg_info
> error: [Errno 13] Permission denied

jhwangbo, 2019-09-07:
It is asking for a permission because you are trying to access a directory that you don't have a permission to work with. This can be due to multiple reasons. Maybe you used sudo when creating your virtualenv. This issue is not related to raisimGym. Please check virtualenv manual to diagnose your problem.

haloted, 2019-09-08:
I followed the installation instruction as posted for nvidia-docker2 and it showed the virtualenv is installed using sudo. However I tried to install virtualenv with a specified python module that is installed locally without sudo: [https://www.youtube.com/watch?v=XhzHitTivWg](url)
And it still showing the same error. My suspicion is in where or how CMake is configured to build the setup.py environment. If I were to reinstall Raisim, how could I make sure that I successfully purged everything? I don't need to reinstall Nvidia or any other dependencies such as Yaml right?

jhwangbo, 2019-09-10:
Hi Ted, you can remove the your install dir $LOCAL_BUILD to uninstall raisim. If you ised cmake to install raisim, it generated install_manifest that lists all the installed files. I highly doubt that your issue is related to the cmake file of raisimgym