leggedrobotics/raisimGym #13: Quadcoptor not reaching the red dot after training.

Aakriti05, 2019-10-01:
[image]

As you can see the image above, after training using the code -  quadrotor_position_tracking.py, the model tends to bring the quadcoptor to a particular location just below the red dot. I have trained the model till 9000 iteration and starting from 1000, it brings to that specific location which is just below the red dot. Can anyone tell me why it is not reaching the red dot exactly and why to a certain location below the red dot?

jhwangbo, 2019-10-01:
Hi Aakriti05, I also observe this. I quickly made an example to show that it is possible to simulate a quadrotor but didn't optimize it. It is also missing a PD controller that I used in my work. Feel free to contribute if you figure out an issue in the example. I'll take a look at it once I have time but not soon. cheers

closed 2019-10-01 by jhwangbo

haloted, 2019-10-02:
Hi Jemin, 
If we want to add the Kp and Kd gain values. Is it as simple as just editting the environment.hpp file at the section where the body virtual torque is produced to include a product with the kp and kd 3 by 3 matrix? Thanks.

jhwangbo, 2019-10-02:
Yes, I recommend checking out the original RAI implementation for the PD controller. It will save you time.

haloted, 2019-10-02:
I'm trying to make a comparison between the source code from bitbucket and the Environment.hpp file Jemin created for the quadrotor.  
To implement the k_p and k_d for rotational matrix, I needed states information such as q_, u_ defined in the C++ source code. 
These are absent from the Environment.hpp file though it is defined as a public variable which should be able to be accessed by any of the methods within the class. There are the obDouble_ and obScaled_ where one is the scaled version of the other, should I be using the scaled ob matrix data or the unscaled data for the PD controller input?
 

```
    //Get current state from q_, u_
    orientation = q_.head(4); //Orientation of quadrotor
    double angle = 2.0 * std::acos(q_(0)); //Angle of rotation

    position = q_.tail(3); //Position of quadrotor
    R_ = Math::MathFunc::quatToRotMat(orientation); //Calculate rotation matrix

    v_I_ = u_.tail(3); //linear velocity
    w_I_ = u_.head(3); //angular velocity
    w_B_ = R_.transpose() * w_I_; //body rates // TODO: u_.head(3) = w_I_

    //Control input from action

    Action actionGenForce;
    actionGenForce = transsThrust2GenForce * actionScale_ * action_t; //Force generated by action
    B_torque = actionGenForce.segment(0, 3); //Moment input
    B_force << 0.0, 0.0, actionGenForce(3); // Thrust vector in {B}

    //Control input from PD stabilization
    double kp_rot = -0.2, kd_rot = -0.06;
    Torque fbTorque_b;

    if (angle > 1e-6)
      fbTorque_b = kp_rot * angle * (R_.transpose() * orientation.tail(3))
          / std::sin(angle) + kd_rot * (R_.transpose() * u_.head(3));
    else
      fbTorque_b = kd_rot * (R_.transpose() * u_.head(3));
    fbTorque_b(2) = fbTorque_b(2) * 0.15; //Lower yaw gains

    B_torque += fbTorque_b; //Sum of torque inputs
    B_force(2) += mass_ * 9.81; //Sum of thrust inputs

    // clip inputs
    Action genForce;
    genForce << B_torque, B_force(2);
    Action thrust = transsThrust2GenForceInv * genForce;    // TODO: This is not exactly thrust: change name
    thrust = thrust.array().cwiseMax(1e-8); // clip for min value
    genForce = transsThrust2GenForce * thrust;
    B_torque = genForce.segment(0, 3);
    B_force(2) = genForce(3);
```
There some undefined classes such as Quaternion, Torque, AngularVelocity etc. which I assume I could just define it as a Vec<3> array, I couldn't find class definitions in the original RAI bucket, could be in the typedef.h which seemed deleted?
All I needed to do is follow the steps here and put the results inside between this block in Environment.hpp?

```
  float step(const Eigen::Ref<EigenVec>& action) final {
    /// action scaling
    thrusts_ = action.cast<double>();
    thrusts_ = thrusts_.cwiseProduct(actionStd_);
    thrusts_ += actionMean_;

    Eigen::Vector4d genForce = transsThrust2GenForce_ * thrusts_;
    Eigen::Vector3d torque_B = genForce.segment(0,3);
    Eigen::Vector3d force_B;
    force_B << 0.0, 0.0, genForce(3);

    /// This part should include a Control Input from PD stabilisation
    
    

    raisim::Vec<3> torque_W, force_W;
    torque_W.e() = rot_.e() * torque_B;
    force_W.e() = rot_.e() * force_B;
```
Also what does the .e() method from the rot_.e() and torque_W.e() used for, I can't find any online references on this method? 
Just a final check before I implement this would be grateful.

jhwangbo, 2019-10-03:
Ted, use a proper IDE like Clion. Then you will easily find answers to most of the questions.

surgeonofdeath, 2019-11-05:
As per the previous replies, I added the PD controller during training time. I implemented the same PD controller as the one in your BitBucket repository. After training with the PD controller, I removed it during testing time. The quadrotor is still not able to reach the red dot. I have attached a few recorded videos for your reference.
During testing: [Testing](https://drive.google.com/open?id=1u10qksZuhwhiPTv9n-wuARan4UXyjIHZ) 
Epoch 3500: [3500](https://drive.google.com/open?id=1dyNCcnXJmy0Gc6Gbh41INGcSH_CY8i-e)
Epoch 3000: [3000](https://drive.google.com/open?id=1MV7ZNI4N2KPslJikdbpDJfq5DnoEjWxq)
Epoch 2500: [2500](https://drive.google.com/open?id=1UPoIRiVwJtZ1XLTV3R58WAW6X4wTI9Nk)
Epoch 2000: [2000](https://drive.google.com/open?id=1HlDJjbbgr7Q9KeEaejuZWGZnHMxw7_3u)

I have changed the camera to the side view so that the error in height can easily be observed.
It looks like there's some sort of steady state error in position. Could you shed some light on this behaviour?