leggedrobotics/raisimGym #14: NaN values in returned reward from the Environment

surgeonofdeath, 2019-10-01:
I am getting NaN values from the environment of the quadcopter. Can you tell me scenarios in which this can happen (Eg: Quadcopter is going out of bounds, quadcopter is rotating too fast etc.)?

TIA

jhwangbo, 2019-10-01:
It happens when the policy becomes extreme and the angular velocity reaches too high values. This can happen since the angular velocity integration cannot be guaranteed to conserve energy. You can set a termination condition to prevent this (e.g., terminate if angular velocity norm is > 20).

closed 2019-10-01 by jhwangbo

surgeonofdeath, 2019-10-01:
Okay, I shall try putting in constraints to stop the episode in extreme cases. I was also looking at the RAI bitbucket repository, and it requires access to the raigraphics_opengl repository. Is it possible to get access to this repository?

Thank you for clarification.

jhwangbo, 2019-10-01:
rai is not managed anymore. I found that raisimGym way of workflow saves a lot of time. So I'll keep raigraphics_opengl closed for now (I had a license issue with the mesh files there). You can still find my original implementation of a quadrotor task in rai.

surgeonofdeath, 2019-10-02:
Okay, I shall look into the quadcopter implementation. 

I looked into the issue of the NaN values and turn out, the `position_` variable has NaN values (in either x, y or z). After a bit more experimentation, I found out that reducing the number of environments to 50 and the number of threads to 1, I managed to avoid getting NaN values entirely. 

Is this because the cumulative of 500 environments * 40 threads was too large value? Could you shed some light on this behavior?

Thank you so much for your help.