leggedrobotics/raisimGym #16: use example in docker  it error Segmentation fault (core dumped)

yonchien, 2019-10-16:
my computer is i7 -8650U , have no gpu. I have tried using docker .but when I run python3 scripts/anymal_blind_locomotion.py. it said Segmentation fault (core dumped). what i can do ?
[image]

jhwangbo, 2019-10-16:
I only made the GPU version. You are welcome to contribute a CPU version ;)

closed 2019-10-16 by jhwangbo

yonchien, 2019-10-16:
what the difference between the GPU and CPU. NOW I use the cpu ,but it said Segmentation fault.  if your code can not use in CPU version

jhwangbo, 2019-10-16:
It uses nvidia-docker. If you normal docker, it will be fine. Check stable-baselines docker files

yonchien, 2019-10-16:
where is stable-baselines docker files