leggedrobotics/raisimGym #17: The normal docker error Segmentation fault (core dumped)

yonchien, 2019-10-17:
I tried yesterday.but still error Segmentation fault (core dumped).
I used the normal docker because I have no GPU. but I tried all examples , it error Segmentation fault (core dumped) and I can not found what's wrong . it occur when I load the raisimOgre
[image]

yonchien, 2019-10-17:
I tried the example cartpole . It still have this question .

jhwangbo, 2019-10-21:
tensorflow is the only place it uses gpu. I am not sure where it can be a problem. For now, I am not supporting cpu docker since it's hard to imagine any use case. If you get it working please contribute by a pull request.

closed 2019-10-21 by jhwangbo