leggedrobotics/raisimGym #18: Option to Pan and Save a Recording during policy testing

haloted, 2019-10-18:
Hi Jemin,
Is there an option to pan and to save a recording in RaisimOgre? 

Cheers,
Ted

jhwangbo, 2019-10-18:
The recording is automatically done. Check the data folder.
You can delete setYawPitchDist function in the step method to freely move around the camera

Cheers

haloted, 2019-10-18:
The recording is done automatically in training at a fixed time. However in testing, there is the option to press the recording button on the panel, that didn't work. 
And I can freely move the camera by deleting the setYawPitchDist function as described.

jhwangbo, 2019-10-18:
can you be more specific? did it give you an error message?

haloted, 2019-10-18:
No it didn't give me an error message. I just couldn't find the video file inside the data folder, only the default.yaml and Environmental.hpp files were present.

jhwangbo, 2019-10-21:
It now saves the video to the install directory which is not optimal. You can call video recording from python side and directly specify the path

closed 2019-10-21 by jhwangbo