leggedrobotics/raisimGym #19: TypeError: savez() argument after ** must be a mapping, not list

yonchien, 2019-10-23:
I tried the example anymal and laikago but only runner a while . then error TypeError: savez() argument after ** must be a mapping, not list .  
[image]

jhwangbo, 2019-10-23:
please copy-paste it so it is searchable

yonchien, 2019-10-23:
Using /usr/local/lib/python3.5/dist-packages
Finished processing dependencies for raisim-gym==0.5.0
/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/dtypes.py:516: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/dtypes.py:517: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/dtypes.py:518: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/dtypes.py:519: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/dtypes.py:520: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/dtypes.py:525: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
/usr/local/lib/python3.5/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:541: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/usr/local/lib/python3.5/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:542: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/usr/local/lib/python3.5/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:543: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/usr/local/lib/python3.5/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:544: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/usr/local/lib/python3.5/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:545: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/usr/local/lib/python3.5/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:550: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
[2019:10:23:14:47:20 OgreVis.hpp:303] Loading RaisimOgre Resource from: /raisim_build/share/raisimOgre/cmake/../rsc
[2019:10:23:14:47:20 OgreVis.hpp:304] Loading Ogre Configuration from: /raisim_build/share/raisimOgre/cmake/../ogre/
WARNING:tensorflow:From /usr/local/lib/python3.5/dist-packages/stable_baselines-2.8.0-py3.5.egg/stable_baselines/common/tf_util.py:57: The name tf.ConfigProto is deprecated. Please use tf.compat.v1.ConfigProto instead.

WARNING:tensorflow:From /usr/local/lib/python3.5/dist-packages/stable_baselines-2.8.0-py3.5.egg/stable_baselines/common/tf_util.py:66: The name tf.Session is deprecated. Please use tf.compat.v1.Session instead.

WARNING:tensorflow:From /usr/local/lib/python3.5/dist-packages/raisim_gym-0.5.0-py3.5-linux-x86_64.egg/raisim_gym/archi/policies.py:112: The name tf.variable_scope is deprecated. Please use tf.compat.v1.variable_scope instead.

WARNING:tensorflow:From /usr/local/lib/python3.5/dist-packages/stable_baselines-2.8.0-py3.5.egg/stable_baselines/common/input.py:25: The name tf.placeholder is deprecated. Please use tf.compat.v1.placeholder instead.

WARNING:tensorflow:From /usr/local/lib/python3.5/dist-packages/raisim_gym-0.5.0-py3.5-linux-x86_64.egg/raisim_gym/archi/policies.py:436: flatten (from tensorflow.python.layers.core) is deprecated and will be removed in a future version.
Instructions for updating:
Use keras.layers.flatten instead.
WARNING:tensorflow:From /usr/local/lib/python3.5/dist-packages/stable_baselines-2.8.0-py3.5.egg/stable_baselines/a2c/utils.py:156: The name tf.get_variable is deprecated. Please use tf.compat.v1.get_variable instead.

[... cut ...]
[libx264 @ 0x559017ea6cc0] mb B  I16..4:  1.7%  0.0%  0.2%  B16..8:  8.2%  1.6%  0.2%  direct: 0.2%  skip:87.9%  L0:50.8% L1:46.4% BI: 2.8%
[libx264 @ 0x559017ea6cc0] coded y,u,v intra: 6.0% 1.4% 3.1% inter: 0.7% 0.1% 0.2%
[libx264 @ 0x559017ea6cc0] i16 v,h,dc,p: 44% 54%  2%  0%
[libx264 @ 0x559017ea6cc0] i4 v,h,dc,ddl,ddr,vr,hd,vl,hu: 25% 28% 26%  4%  4%  4%  4%  3%  3%
[libx264 @ 0x559017ea6cc0] Weighted P-Frames: Y:0.0% UV:0.0%
[libx264 @ 0x559017ea6cc0] ref P L0: 60.0% 40.0%
[libx264 @ 0x559017ea6cc0] ref B L0: 80.5% 19.5%
[libx264 @ 0x559017ea6cc0] ref B L1: 91.2%  8.8%
[libx264 @ 0x559017ea6cc0] kb/s:1332.09

jhwangbo, 2019-10-23:
is this on CPU docker?

yonchien, 2019-10-23:
no,it is not  a docker , I use GPU RTX2060 + CUDA 10.1

jhwangbo, 2019-10-23:
which file did you run?

yonchien, 2019-10-23:
python3 setup.py install --CMAKE_PREFIX_PATH /raisim_build --env anymal && python3 scripts/anymal_blind_locomotion.py

jhwangbo, 2019-10-23:
did you modify the code? currently, there is no saving in the examples but you have this line

py3.5.egg/stable_baselines/common/save_util.py", line 150, in params_to_bytes
np.savez(byte_file, **params)

yonchien, 2019-10-23:
I did not modify the code . but i use python3.5

jhwangbo, 2019-10-23:
Thank you for reporting. I figured out it does not run with the latest stable-baselines. I'll update asap

jhwangbo, 2019-10-23:
changes have been pushed

closed 2019-10-23 by jhwangbo

yonchien, 2019-10-23:
what stable-bashlines do you use

jhwangbo, 2019-10-23:
2.7 before. Now I made it only work with 2.8 or above