leggedrobotics/raisimGym #26: Quadrotor having a steady-state error and not reaching red dot.

surgeonofdeath, 2019-11-13:
As per the previous replies, I added the PD controller during training time. I implemented the same PD controller as the one in your BitBucket repository. After training with the PD controller, I removed it during testing time. The quadrotor is still not able to reach the red dot. I have attached a few recorded videos for your reference.
During testing: [Testing](https://drive.google.com/open?id=1u10qksZuhwhiPTv9n-wuARan4UXyjIHZ) 
Epoch 3500: [3500](https://drive.google.com/open?id=1dyNCcnXJmy0Gc6Gbh41INGcSH_CY8i-e)
Epoch 3000: [3000](https://drive.google.com/open?id=1MV7ZNI4N2KPslJikdbpDJfq5DnoEjWxq)
Epoch 2500: [2500](https://drive.google.com/open?id=1UPoIRiVwJtZ1XLTV3R58WAW6X4wTI9Nk)
Epoch 2000: [2000](https://drive.google.com/open?id=1HlDJjbbgr7Q9KeEaejuZWGZnHMxw7_3u)

I have changed the camera to the side view so that the error in height can easily be observed.
It looks like there's some sort of steady state error in position. Could you shed some light on this behaviour?
Did you have this problem during the writing of your paper (Quadrotor control using reinforcement learning)?

jhwangbo, 2019-11-15:
Why do you remove the PD controller during the test time? It is meant to work in parallel with the neural net.

As written on the paper, my implementation created about 3~4cm or constant position offset. This is due to the fact that we are using a function approximator for the policy. I also used my custom algorithm as written in the paper and it performed better than PPO for this task.

surgeonofdeath, 2019-11-15:
I did further testing with the PD controller working in parallel with the neural net. I also observed a similar offset in that scenario.

Did your custom algorithm have a lesser offset as compared to PPO? If so, I will go ahead and implement the same. 

Thanks.

jhwangbo, 2019-11-15:
I did not compare them in terms of steady-state offset. I compared them with the cost in the paper. but they must correlate.

My algorithm is already implemented in the original RAI repo (C++). I don't have time to manage it and that's why it is not transferred to raisimGym. You can take a look there.

closed 2019-11-21 by jhwangbo