leggedrobotics/raisimGym #28: Segmentation fault when run train in docker

iskanderuse, 2019-12-02:
We try to run train process in docker in Azure with no gpu and get "Segmentation fault".

I tryed [https://github.com/leggedrobotics/raisimGym/issues/4], it doesn't work.
An error occurs on this line "from _raisim_gym import RaisimGymEnv" of python file with example.

jhwangbo, 2019-12-02:
this seems like tensorflow version issue. RaisimGym uses 1.14 but the current docker script calls `pip3 install tensorflow-gpu` which will install tensorflow 2.0 by default. I am running the script now to confirm.

jhwangbo, 2019-12-02:
There were two issues:
1. pybind11 error which caused the segfault. I believe this was a bug from pybind11 and it was solved by upgrading the pybind11 to 2.4.3
2. tensorflow version error. now it is set to 1.14
the changes are reflected in the latest commit.

closed 2019-12-02 by jhwangbo