leggedrobotics/raisimGym #33: Problem in visualization

Fleer-code, 2019-12-25:
Hello, when I use kinova_stand.urdf to do Reinforcement Learning, I met the following errors. How can I fix these? 
`[RAISIM_GYM] Visualizing in RaiSimOgre
*** buffer overflow detected ***: python3 terminated
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7fc2ee2357e5]
/lib/x86_64-linux-gnu/libc.so.6(__fortify_fail+0x5c)[0x7fc2ee2d715c]
/lib/x86_64-linux-gnu/libc.so.6(+0x117160)[0x7fc2ee2d5160]
/lib/x86_64-linux-gnu/libc.so.6(+0x1166c9)[0x7fc2ee2d46c9]
/lib/x86_64-linux-gnu/libc.so.6(_IO_default_xsputn+0x80)[0x7fc2ee2396b0]
/lib/x86_64-linux-gnu/libc.so.6(+0x5225a)[0x7fc2ee21025a]
/lib/x86_64-linux-gnu/libc.so.6(_IO_vfprintf+0x1f49)[0x7fc2ee20d0b9]
/lib/x86_64-linux-gnu/libc.so.6(__vsprintf_chk+0x84)[0x7fc2ee2d4754]
/lib/x86_64-linux-gnu/libc.so.6(__sprintf_chk+0x7d)[0x7fc2ee2d46ad]
/usr/local/lib/python3.5/dist-packages/raisim_gym-0.5.0-py3.5-linux-x86_64.egg/_raisim_gym.cpython-35m-x86_64-linux-gnu.so(+0x19abe)[0x7fc2616b8abe]
/home/wzc/raisim_build/lib/libraisimOgre.so(_ZN6raisim7OgreVis12frameStartedERKN4Ogre10FrameEventE+0xb3)[0x7fc26146d383]
/home/wzc/raisim_build/lib/libOgreMain.so.1.12.1(_ZN4Ogre4Root17_fireFrameStartedERNS_10FrameEventE+0x98)[0x7fc2608acd48]
/home/wzc/raisim_build/lib/libOgreMain.so.1.12.1(_ZN4Ogre4Root17_fireFrameStartedEv+0x31)[0x7fc2608ad301]
/home/wzc/raisim_build/lib/libOgreMain.so.1.12.1(_ZN4Ogre4Root14renderOneFrameEv+0x9)[0x7fc2608ad449]
/home/wzc/raisim_build/lib/libraisimOgre.so(_ZN6raisim7OgreVis14renderOneFrameEv+0x28b)[0x7fc26147098b]
/usr/local/lib/python3.5/dist-packages/raisim_gym-0.5.0-py3.5-linux-x86_64.egg/_raisim_gym.cpython-35m-x86_64-linux-gnu.so(+0x1f57d)[0x7fc2616be57d]
/usr/local/lib/python3.5/dist-packages/raisim_gym-0.5.0-py3.5-linux-x86_64.egg/_raisim_gym.cpython-35m-x86_64-linux-gnu.so(+0x1f939)[0x7fc2616be939]
/usr/local/lib/python3.5/dist-packages/raisim_gym-0.5.0-py3.5-linux-x86_64.egg/_raisim_gym.cpython-35m-x86_64-linux-gnu.so(+0x15eed)[0x7fc2616b4eed]
/usr/local/lib/python3.5/dist-packages/raisim_gym-0.5.0-py3.5-linux-x86_64.egg/_raisim_gym.cpython-35m-x86_64-linux-gnu.so(+0x26211)[0x7fc2616c5211]
python3(PyCFunction_Call+0x77)[0x4e1307]
python3(PyEval_EvalFrameEx+0x5a4b)[0x535fcb]
python3[0x539f5f]
python3(PyEval_EvalFrameEx+0x5122)[0x5356a2]
python3[0x539a13]
python3(PyEval_EvalFrameEx+0x5122)[0x5356a2]
python3[0x539f5f]
python3(PyEval_EvalFrameEx+0x5122)[0x5356a2]
python3[0x539a13]
python3(PyEval_EvalCode+0x1f)[0x53a6cf]
python3[0x6292c2]
python3(PyRun_FileExFlags+0x9a)[0x62b76a]
python3(PyRun_SimpleFileExFlags+0x1bc)[0x62bf5c]
python3(Py_Main+0x456)[0x63d506]
python3(main+0xe1)[0x4cfd11]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7fc2ee1de830]
python3(_start+0x29)[0x5d36e9]
======= Memory map: ========
00400000-007a7000 r-xp 00000000 08:06 4559                               /usr/bin/python3.5
009a6000-009a8000 r--p 003a6000 08:06 4559                               /usr/bin/python3.5
009a8000-00a3f000 rw-p 003a8000 08:06 4559                               /usr/bin/python3.5
00a3f000-00a70000 rw-p 00000000 00:00 0 
013a2000-0dec2000 rw-p 00000000 00:00 0                                  [heap]
10000000-20000000 ---p 00000000 00:00 0 
40968000-4096a000 r-xs 00000000 08:06 439619                             /tmp/.glK8WZll (deleted)
40a89000-40b2a000 rw-p 00000000 00:00 0 
200000000-400200000 ---p 00000000 00:00 0 
10000000000-10200000000 ---p 00000000 00:00 0 
7fc050000000-7fc050021000 rw-p 00000000 00:00 0 
7fc050021000-7fc054000000 ---p 00000000 00:00 0 
7fc054000000-7fc054052000 rw-p 00000000 00:00 0 
7fc054052000-7fc058000000 ---p 00000000 00:00 0 
7fc058000000-7fc05804c000 rw-p 00000000 00:00 0 
7fc05804c000-7fc05c000000 ---p 00000000 00:00 0 
7fc05c000000-7fc05c54e000 rw-p 00000000 00:00 0 
7fc05c54e000-7fc060000000 ---p 00000000 00:00 0 
7fc060000000-7fc060056000 rw-p 00000000 00:00 0 
7fc060056000-7fc064000000 ---p 00000000 00:00 0 
7fc0647fd000-7fc0647fe000 ---p 00000000 00:00 0 
7fc0647fe000-7fc064ffe000 rw-p 00000000 00:00 0 
7fc0657ff000-7fc065800000 ---p 00000000 00:00 0 
7fc065800000-7fc066000000 rw-p 00000000 00:00 0 
7fc066000000-7fc068000000 rw-p 00000000 00:00 0 
7fc068000000-7fc068021000 rw-p 00000000 00:00 0 
7fc068021000-7fc06c000000 ---p 00000000 00:00 0 
7fc06d7fb000-7fc06d7fc000 ---p 00000000 00:00 0 
7fc06d7fc000-7fc06dffc000 rw-p 00000000 00:00 0 
[... 586 lines cut ...]
  libswscale      3.  1.101 /  3.  1.101
  libswresample   1.  2.101 /  1.  2.101
  libpostproc    53.  3.100 / 53.  3.100
Input #0, rawvideo, from 'pipe:':
  Duration: N/A, start: 0.000000, bitrate: N/A
    Stream #0:0: Video: rawvideo (RGB[24] / 0x18424752), rgb24, 1800x1027, -2076647 kb/s, 50 tbr, 50 tbn, 50 tbc
No pixel format specified, yuv444p for H.264 encoding chosen.
Use -pix_fmt yuv420p for compatibility with outdated media players.
[libx264 @ 0x1c6fbe0] using cpu capabilities: MMX2 SSE2Fast SSSE3 SSE4.2 AVX FMA3 AVX2 LZCNT BMI2
[libx264 @ 0x1c6fbe0] profile High 4:4:4 Predictive, level 4.2, 4:4:4 8-bit
[libx264 @ 0x1c6fbe0] 264 - core 148 r2643 5c65704 - H.264/MPEG-4 AVC codec - Copyleft 2003-2015 - http://www.videolan.org/x264.html - options: cabac=1 ref=2 deblock=1:0:0 analyse=0x3:0x113 me=hex subme=6 psy=1 psy_rd=1.00:0.00 mixed_ref=1 me_range=16 chroma_me=1 trellis=1 8x8dct=1 cqm=0 deadzone=21,11 fast_pskip=1 chroma_qp_offset=4 threads=32 lookahead_threads=5 sliced_threads=0 nr=0 decimate=1 interlaced=0 bluray_compat=0 constrained_intra=0 bframes=3 b_pyramid=2 b_adapt=1 b_bias=0 direct=1 weightb=1 open_gop=0 weightp=1 keyint=250 keyint_min=25 scenecut=40 intra_refresh=0 rc_lookahead=30 rc=crf mbtree=1 crf=21.0 qcomp=0.60 qpmin=0 qpmax=69 qpstep=4 ip_ratio=1.40 aq=1:1.00
Output #0, mp4, to '/home/wzc/raisim_workspace/raisimGym/raisim_gym/env/env/cartpole/../cartpole/data/Cartpole_tutorial/2019-12-25-12-10-16/0_0.mp4':
  Metadata:
    encoder         : Lavf56.40.101
    Stream #0:0: Video: h264 (libx264) ([33][0][0][0] / 0x0021), yuv444p, 1800x1027, q=-1--1, 50 fps, 12800 tbn, 50 tbc
    Metadata:
      encoder         : Lavc56.60.100 libx264
Stream mapping:
  Stream #0:0 -> #0:0 (rawvideo (native) -> h264 (libx264))
[rawvideo @ 0x1c6e620] Invalid buffer size, packet size 65536 < expected frame_size 5545800
Error while decoding stream #0:0: Invalid argument
frame=    0 fps=0.0 q=0.0 Lsize=       0kB time=00:00:00.00 bitrate=N/A    
video:0kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown
Conversion failed!
Aborted (core dumped)`

jhwangbo, 2019-12-25:
Which code did you run? Is this raisimGym style or raisimPy style? When do you get this error?

Fleer-code, 2019-12-25:
I ran the code provided in the raisimGym tutorial. However I changed the `cartpole.urdf` to `kinova_stand.urdf` and made some corresponding changes, too. I compiled the project using setup.py successfully, but got the errors after I run kinova.py  which content is the same as cartpole_tutorpal.py.

jhwangbo, 2019-12-26:
Can you run the debug app with Valgrind and check if you have anything suspicious? With '--no-render' flag, you shoudnt get any warnings.

closed 2020-07-16 by Fleer-code