leggedrobotics/raisimGym #34: Relationship to ANYmal science paper

hzyjerry, 2020-02-17:
Hi,

Thanks for the effort building and releasing raisimGym! I am wondering how much does the ANYmal environment in raisimGym resemble the [one used for ANYmal science paper](https://github.com/junja94/anymal_science_robotics_supplementary)? It seems that the cost function in raisimGym is a simplified version with 2 terms. Are the policies trained from raisim gym still deployable on the physical ANYmal robot?

Thanks!

jhwangbo, 2020-02-17:
Unfortunately, the examples in raisim gym are for AI benchmark, not for deployment. The anymal example is missing actuator model and many necessary terms in the reward function. I just wanted to share how fast it can train if it uses simple tasks that other people use.

Implementing the Science Robotics version shouldn't be that hard in raisimGym. I am exactly doing that for my current project. The difficult part is getting data from your actuator to build an actuator network.

hzyjerry, 2020-02-17:
@jhwangbo thanks for the swift response. 

It totally makes sense for the use case of raisim gym. It seems that [most terms in the reward function](https://github.com/junja94/anymal_science_robotics_supplementary/blob/master/environments/task/include/quadrupedLocomotion/ANYmal_minimal.hpp#L982) could be implemented without actuator model. I'm working on a project for reward design in RL & optimal control. Basically trying to eliminate the manual effort for finding the coefficients for good reward functions. This seems to be a perfect use case.

Excited to hear you're working on Implementing the Science Robotics version for raisimGym. Would be a solid contribution to the community!

closed 2020-02-17 by hzyjerry