leggedrobotics/raisimGym #39: Issues in Running the examples

kashifzr, 2020-03-26:
I was using raisimGym on ubuntu 16.04 without GPU, where its work fine without such issue give below, Now i am using GPU on ubuntu 18.04 where i compile all the stuff required to run the raisimGym Now after two successive attempts to run the code, i am getting the same issue again and again. the file  libOgreRTShaderSystem.so.1.12.2 is found the structure below. But the error is says No such file or directory

<pre>np_resource = np.dtype([(&quot;resource&quot;, np.ubyte, 1)])
Traceback (most recent call last):
  File &quot;scripts/anymal_blind_locomotion.py&quot;, line 3, in &lt;module&gt;
    from raisim_gym.env.env.ANYmal import __ANYMAL_RESOURCE_DIRECTORY__ as __RSCDIR__
<pre>File &quot;/home/abc/.local/lib/python3.6/site-packages/raisim_gym-0.5.0-py3.6-linux-x86_64.egg/raisim_gym/env/env/ANYmal/__init__.py&quot;, line 2, in &lt;module&gt;
    from _raisim_gym import *
ImportError: libOgreRTShaderSystem.so.1.12.2: cannot open shared object file: No such file or directory
</pre>

**LOCAL BUILD folder with all the data which required to run the code** 

<pre>│   │   ├── heap<font color="#EF2929"><b>so</b></font>rt.h
│   │   │   ├── OgreSGTechniqueRe<font color="#EF2929"><b>so</b></font>lverListener.h
│   │   ├── OgreRe<font color="#EF2929"><b>so</b></font>urceBackgroundQueue.h
│   │   ├── OgreRe<font color="#EF2929"><b>so</b></font>urceGroupManager.h
│   │   ├── OgreRe<font color="#EF2929"><b>so</b></font>urce.h
│   │   ├── OgreRe<font color="#EF2929"><b>so</b></font>urceManager.h
│   │   │   │   ├── OgreEllip<font color="#EF2929"><b>so</b></font>idEmitterFactory.h
│   │   │   │   ├── OgreEllip<font color="#EF2929"><b>so</b></font>idEmitter.h
│   │   │   │   ├── OgreHollowEllip<font color="#EF2929"><b>so</b></font>idEmitterFactory.h
│   │   │   │   ├── OgreHollowEllip<font color="#EF2929"><b>so</b></font>idEmitter.h
│   │   │   ├── OgreShaderCGProgramProces<font color="#EF2929"><b>so</b></font>r.h
│   │   │   ├── OgreShaderGLSLESProgramProces<font color="#EF2929"><b>so</b></font>r.h
│   │   │   ├── OgreShaderGLSLProgramProces<font color="#EF2929"><b>so</b></font>r.h
│   │   │   ├── OgreShaderHLSLProgramProces<font color="#EF2929"><b>so</b></font>r.h
│   │   │   ├── OgreShaderProgramProces<font color="#EF2929"><b>so</b></font>r.h
│   │       ├── OgreVolumeI<font color="#EF2929"><b>so</b></font>Surface.h
│   │       ├── OgreVolumeI<font color="#EF2929"><b>so</b></font>SurfaceMC.h
│   │       ├── OgreVolumeI<font color="#EF2929"><b>so</b></font>SurfaceTablesMC.h
│   ├── libassimp.<font color="#EF2929"><b>so</b></font> -&gt; libassimp.<font color="#EF2929"><b>so</b></font>.4
│   ├── libassimp.<font color="#EF2929"><b>so</b></font>.4 -&gt; libassimp.<font color="#EF2929"><b>so</b></font>.4.1.0
│   ├── libassimp.<font color="#EF2929"><b>so</b></font>.4.1.0
│   ├── libOgreBites.<font color="#EF2929"><b>so</b></font> -&gt; libOgreBites.<font color="#EF2929"><b>so</b></font>.1.12.2
│   ├── libOgreBites.<font color="#EF2929"><b>so</b></font>.1.12.2
│   ├── libOgreHLMS.<font color="#EF2929"><b>so</b></font> -&gt; libOgreHLMS.<font color="#EF2929"><b>so</b></font>.1.12.2
│   ├── libOgreHLMS.<font color="#EF2929"><b>so</b></font>.1.12.2
│   ├── libOgreMain.<font color="#EF2929"><b>so</b></font> -&gt; libOgreMain.<font color="#EF2929"><b>so</b></font>.1.12.2
│   ├── libOgreMain.<font color="#EF2929"><b>so</b></font>.1.12.2
│   ├── libOgreMeshLodGenerator.<font color="#EF2929"><b>so</b></font> -&gt; libOgreMeshLodGenerator.<font color="#EF2929"><b>so</b></font>.1.12.2
│   ├── libOgreMeshLodGenerator.<font color="#EF2929"><b>so</b></font>.1.12.2
│   ├── libOgreOverlay.<font color="#EF2929"><b>so</b></font> -&gt; libOgreOverlay.<font color="#EF2929"><b>so</b></font>.1.12.2
│   ├── libOgreOverlay.<font color="#EF2929"><b>so</b></font>.1.12.2
│   ├── libOgrePaging.<font color="#EF2929"><b>so</b></font> -&gt; libOgrePaging.<font color="#EF2929"><b>so</b></font>.1.12.2
│   ├── libOgrePaging.<font color="#EF2929"><b>so</b></font>.1.12.2
│   ├── libOgreProperty.<font color="#EF2929"><b>so</b></font> -&gt; libOgreProperty.<font color="#EF2929"><b>so</b></font>.1.12.2
│   ├── libOgreProperty.<font color="#EF2929"><b>so</b></font>.1.12.2
│   ├── libOgreRTShaderSystem.<font color="#EF2929"><b>so</b></font> -&gt; libOgreRTShaderSystem.<font color="#EF2929"><b>so</b></font>.1.12.2
│   ├── libOgreRTShaderSystem.<font color="#EF2929"><b>so</b></font>.1.12.2
│   ├── libOgreTerrain.<font color="#EF2929"><b>so</b></font> -&gt; libOgreTerrain.<font color="#EF2929"><b>so</b></font>.1.12.2
│   ├── libOgreTerrain.<font color="#EF2929"><b>so</b></font>.1.12.2
│   ├── libOgreVolume.<font color="#EF2929"><b>so</b></font> -&gt; libOgreVolume.<font color="#EF2929"><b>so</b></font>.1.12.2
│   ├── libOgreVolume.<font color="#EF2929"><b>so</b></font>.1.12.2
│   ├── libpng16.<font color="#EF2929"><b>so</b></font>
│   ├── libpng16.<font color="#EF2929"><b>so</b></font>.16
│   ├── libpng16.<font color="#EF2929"><b>so</b></font>.16.38.git
│   ├── libpng.<font color="#EF2929"><b>so</b></font>
│   ├── libraisimODE.<font color="#EF2929"><b>so</b></font>
│   ├── libraisimOgre.<font color="#EF2929"><b>so</b></font>
│   ├── libraisim.<font color="#EF2929"><b>so</b></font>
│   │   ├── Codec_FreeImage.<font color="#EF2929"><b>so</b></font> -&gt; Codec_FreeImage.<font color="#EF2929"><b>so</b></font>.1.12.2
│   │   ├── Codec_FreeImage.<font color="#EF2929"><b>so</b></font>.1.12.2
│   │   ├── Codec_STBI.<font color="#EF2929"><b>so</b></font> -&gt; Codec_STBI.<font color="#EF2929"><b>so</b></font>.1.12.2
│   │   ├── Codec_STBI.<font color="#EF2929"><b>so</b></font>.1.12.2
│   │   ├── Plugin_BSPSceneManager.<font color="#EF2929"><b>so</b></font> -&gt; Plugin_BSPSceneManager.<font color="#EF2929"><b>so</b></font>.1.12.2
│   │   ├── Plugin_BSPSceneManager.<font color="#EF2929"><b>so</b></font>.1.12.2
│   │   ├── Plugin_DotScene.<font color="#EF2929"><b>so</b></font> -&gt; Plugin_DotScene.<font color="#EF2929"><b>so</b></font>.1.12.2
│   │   ├── Plugin_DotScene.<font color="#EF2929"><b>so</b></font>.1.12.2
│   │   ├── Plugin_OctreeSceneManager.<font color="#EF2929"><b>so</b></font> -&gt; Plugin_OctreeSceneManager.<font color="#EF2929"><b>so</b></font>.1.12.2
│   │   ├── Plugin_OctreeSceneManager.<font color="#EF2929"><b>so</b></font>.1.12.2
│   │   ├── Plugin_OctreeZone.<font color="#EF2929"><b>so</b></font> -&gt; Plugin_OctreeZone.<font color="#EF2929"><b>so</b></font>.1.12.2
│   │   ├── Plugin_OctreeZone.<font color="#EF2929"><b>so</b></font>.1.12.2
│   │   ├── Plugin_ParticleFX.<font color="#EF2929"><b>so</b></font> -&gt; Plugin_ParticleFX.<font color="#EF2929"><b>so</b></font>.1.12.2
│   │   ├── Plugin_ParticleFX.<font color="#EF2929"><b>so</b></font>.1.12.2
│   │   ├── Plugin_PCZSceneManager.<font color="#EF2929"><b>so</b></font> -&gt; Plugin_PCZSceneManager.<font color="#EF2929"><b>so</b></font>.1.12.2
│   │   ├── Plugin_PCZSceneManager.<font color="#EF2929"><b>so</b></font>.1.12.2
│   │   ├── RenderSystem_GL3Plus.<font color="#EF2929"><b>so</b></font> -&gt; RenderSystem_GL3Plus.<font color="#EF2929"><b>so</b></font>.1.12.2
│   │   ├── RenderSystem_GL3Plus.<font color="#EF2929"><b>so</b></font>.1.12.2
│   │   ├── RenderSystem_GL.<font color="#EF2929"><b>so</b></font> -&gt; RenderSystem_GL.<font color="#EF2929"><b>so</b></font>.1.12.2
│   │   └── RenderSystem_GL.<font color="#EF2929"><b>so</b></font>.1.12.2
    │   │   │   │   │   ├── i<font color="#EF2929"><b>so</b></font>surf.cg
    │   │   │   │   │   ├── I<font color="#EF2929"><b>so</b></font>surfFS.glsl
    │   │   │   │   │   ├── I<font color="#EF2929"><b>so</b></font>surfGS.glsl
    │   │   │   │   │   ├── I<font color="#EF2929"><b>so</b></font>surfVS.glsl
    │   │   │   │   │   ├── i<font color="#EF2929"><b>so</b></font>surf.hlsl
    │   │   │   │   ├── I<font color="#EF2929"><b>so</b></font>Surf.material
    │   │   │       ├── cur<font color="#EF2929"><b>so</b></font>r.png
    │   │   │   ├── thumb_i<font color="#EF2929"><b>so</b></font>surf.png
    │   ├── re<font color="#EF2929"><b>so</b></font>urces.cfg
        │   └── re<font color="#EF2929"><b>so</b></font>urces.cfg
            ├── ogreRe<font color="#EF2929"><b>so</b></font>urces
            │       │   │   │   ├── i<font color="#EF2929"><b>so</b></font>surf.cg
            │       │   │   │   ├── I<font color="#EF2929"><b>so</b></font>surfFS.glsl
            │       │   │   │   ├── I<font color="#EF2929"><b>so</b></font>surfGS.glsl
            │       │   │   │   ├── I<font color="#EF2929"><b>so</b></font>surfVS.glsl
            │       │   │       ├── i<font color="#EF2929"><b>so</b></font>surf.hlsl
            │       │       ├── cur<font color="#EF2929"><b>so</b></font>r.png
            │           ├── thumb_i<font color="#EF2929"><b>so</b></font>surf.png
</pre>

jhwangbo, 2020-03-26:
Hi kashifzr, I think this is nothing to do with gpu. Did you set LD_LIBRARY_PATH? you have to add $LOCAL_BUILD/lib to your LD_LIBRARY_PATH

kashifzr, 2020-03-26:
Honestly I never did before, actually I am never know where to add this path while installation. I did manual installation even installing the raisimGym first time. All I know is to add the path file Bashrc File after building code