leggedrobotics/raisimGym #43: turn off the visualization

ZQX323, 2020-06-08:
Hello,

Thanks for your Repos. I have a question, I want to run raisimGym in a remote machine, so I can not visualize it in remote machine.  Errors is as follows:

[2020:06:08:12:50:32 OgreVis.hpp:342] Loading RaisimOgre Resources from: 
[2020:06:08:12:50:32 OgreVis.hpp:343] Loading OGRE Configurations from: /root/raisim_build/share/raisimOgre/cmake/../ogre/
Segmentation fault (core dumped)

Where can I find the config file to turn raisimOgre off such that there is no error?

jhwangbo, 2020-06-08:
When you instantiate the environment, the constructor takes the location of the config file. For the anymal example, you can find it here: https://github.com/leggedrobotics/raisimGym/blob/master/raisim_gym/env/env/ANYmal/ANYmal_bear2/default_cfg.yaml

ZQX323, 2020-06-08:
> When you instantiate the environment, the constructor takes the location of the config file. For the anymal example, you can find it here: https://github.com/leggedrobotics/raisimGym/blob/master/raisim_gym/env/env/ANYmal/ANYmal_bear2/default_cfg.yaml

It works, thanks for your help!

ZQX323, 2020-06-16:
Hello Jemin,

Sorry to bother you again. It seems Anymal can only be controlled by 12 joint. Is there a function that can direct control the feet of Anymal? For example, direct move the endpoint of one leg from(0, 0, 0) to (1, 1, 0).

closed 2020-07-09 by ZQX323