leggedrobotics/raisimGym #44: Error when use step function in class RaisimGymVecEnv

ZQX323, 2020-07-09:
Hi,

Thanks for sharing this. I want to add td3 from stable-baselines in the algo. When the program execute self.env.step, it cause follow error, I think the input of the function is correct. So it's very confusing. Thanks in advance.

```
Traceback (most recent call last):
  File "scripts/anymal_blind_locomotion_td3.py", line 46, in <module>
    model.learn(total_timesteps=50000, log_interval=10)
  File "/home/zqx/anaconda3/envs/anymal/lib/python3.6/site-packages/raisim_gym-0.5.0-py3.6-linux-x86_64.egg/raisim_gym/algo/td3/td3.py", line 331, in learn
    new_obs, reward, done, info = self.env.step(rescaled_action, visualize=True)
  File "/home/zqx/anaconda3/envs/anymal/lib/python3.6/site-packages/raisim_gym-0.5.0-py3.6-linux-x86_64.egg/raisim_gym/env/RaisimGymVecEnv.py", line 30, in step
    self.wrapper.testStep(action, self._observation, self._reward, self._done, self._extraInfo)
TypeError: testStep(): incompatible function arguments. The following argument types are supported:
    1. (self: _raisim_gym.RaisimGymEnv, arg0: numpy.ndarray[float32[m, n], flags.writeable, flags.c_contiguous], arg1: numpy.ndarray[float32[m, n], flags.writeable, flags.c_contiguous], arg2: numpy.ndarray[float32[m, 1], flags.writeable], arg3: numpy.ndarray[bool[m, 1], flags.writeable], arg4: numpy.ndarray[float32[m, n], flags.writeable, flags.c_contiguous]) -> None

Invoked with: <_raisim_gym.RaisimGymEnv object at 0x7f85b3062848>, array([[ 0.08874022,  0.11524627,  0.08672155, -0.1041653 ,  0.08133964,
        -0.54812552,  0.44476833,  0.3027389 , -0.27096661,  0.6745773 ,
        -0.34092081,  0.11477388]]), array([[ 0.02584356,  0.04644538,  0.5113571 , -0.0745747 , -0.07252908,
         1.4247787 ,  0.04175732, -0.04320183, -0.00686281, -0.10704116,
        -0.38630995,  0.60385   ,  0.00182454,  0.44784692, -0.6802453 ,
         0.05022758,  0.47194725, -0.6429743 ,  0.00186013,  0.03102602,
         0.03337318,  0.0835087 , -0.05459327, -0.08109044,  0.45095894,
         0.3529083 ,  0.0037378 ,  0.529099  , -0.2433785 , -0.02164587,
        -0.53081214,  0.21880619, -0.01959409, -0.49592617, -0.27816093,
         0.01029105,  0.71      ,  0.26      , -0.04309561,  0.46035215,
        -0.26055574, -0.04309561, -0.46035215,  0.26055574, -0.04309561,
        -0.46035215, -0.26055574, -0.04309561,  0.        ,  1.        ,
         1.        ,  1.        ,  1.        ,  1.        ,  0.71      ,
         0.26      , -0.04309561,  0.02132099,  0.02662646,  0.11978799,
        -0.01284828, -0.06666117,  0.12672742, -0.07316593,  0.03450989,
         0.0056312 , -0.07493337, -0.00926625,  0.0332063 ]],
      dtype=float32), array([-0.34402055], dtype=float32), array([False]), array([[0.5113571, 0.       ]], dtype=float32)

```

jhwangbo, 2020-07-10:
It is saying that you called the function with a wrong set of arguments. Is the first array float32? I am not very experienced in python so I can't tell but make sure that they all have the correct data type. The rest seems ok to me too.

ZQX323, 2020-07-10:
yes, when I implement self.env.step(action, visualize=True). 

If action is array([-0.17920911, -0.02736158,  0.37157712,  0.03743708, -0.7466227 ,
       -0.01556192, -0.03418474,  0.19078933, -0.27589707,  0.1042255 ,
       -0.49273024, -0.05912861]) 

it will cause error.

but when action is array([ 0.09762701,  0.43037874,  0.20552675,  0.08976637, -0.1526904 ,
        0.29178822, -0.12482557,  0.78354603,  0.92732555, -0.23311697,
        0.5834501 ,  0.05778984], dtype=float32) 

with dtype=float32, there is no error. But I think both of them should be OK. I am not familiar with pybind, would you mind to tell me which c++ file will be implemented after step/testStep function of class RaisimGymEnv in _raisim_gym.py is called? I want to check it's compatibility with different input.

ZQX323, 2020-07-10:
seems step function only support float32 not float64

jhwangbo, 2020-07-10:
yes, that's right. but this should be a good feature. The type conversion is costly and this helps you remove it ;)

closed 2020-07-10 by jhwangbo