leggedrobotics/raisimGym #45: Question on .pkl file saved after the training.

Fleer-code, 2020-07-16:
Hi,

I have a question on the content of .pkl file saved after the training of anymal_blind_locomotion.py. I used the parameter 
`policy_kwargs=dict(net_arch=[dict(pi=[120, 160], vf=[130, 140])]),`   in PPO2.load(). And when I look into the weights saved in .pkl. I found the shape of the weights are as follows,
(a): 34x120
(b): 120
(c): 34*130
(d): 130
(e): 120x160
(f): 160
(g): 130x140
(h): 140
(i): 140x1
(j): 1
(k): 160x12
(l): 12
(m): 1x12
(n): 140x12
(o): 12
After I read the code, I learned that (a)(b)(e)(f)(k)(l) are weights of the policy network and (c)(d)(g)(h)(i)(j) are weights of the value function  network. But I still confused about the function of (m)(n)(o). According to the dimensions of these arrays, (n) and (o)
 seems to be the weights and bias connected to the last hidden layer of the value function  network, and (m) seems to be connected to the output layer of the vf network. All the weights are extracted by "self.sess.all_trainable_variables()". However. I have no idea what these weights are used for. I wonder if you know the meaning of these weights.

Thanks.

jhwangbo, 2020-07-16:
Standard deviation ;)

Fleer-code, 2020-07-16:
So the (n) and (o) are weights and bias for calculating the Std of the action, and they are connected to the the last hidden layer of the value function network. I am still confused about the usage of (m), it is a 2d array and probably a weight matrix (because the bias are often vectors like (b) and (f).).

jhwangbo, 2020-07-16:
Can't you check the name?

Fleer-code, 2020-07-18:
I read the code again, but it doesn't help. Due to the mechanism of Tensorflow the setup procedure of tensors are in different code files, it is hard to check the name of these variables and I still don't know the role of (o). It seems, in the .pkl, there are 8 lists of weights and 7 lists of bias. Why they are not equal?

Fleer-code, 2020-07-18:
I found the names of the weights. (n) is q_weights and (o) is q_bias. Both of them seems not used in ppo algorithm. Thank you.

closed 2020-07-18 by Fleer-code