leggedrobotics/raisimGym #48: license is expired

app1ep1e, 2020-09-02:
what i input :
python3 scripts/anymal_blind_locomotion.py

what i get some future warning and : 
[2020:09:02:11:52:14 license.hpp:15] THIS LICENSE IS EXPIRED. PLEASE GET A NEW LICENSE

it is everything ok two weeks ago, 
what should i do to solve this problem?

jhwangbo, 2020-09-02:
please pull raisimlib again. We will distribute raisim on a different channel. Before then, we will prolong the license in raisimlib github repo

app1ep1e, 2020-09-02:
Thank you so much for your fast reply. it works.

closed 2020-09-02 by app1ep1e

kashifzr, 2020-09-06:
For my understanding pull means download raisimLib again from source directory and compile the software,
Its does not work. Could some one explain the PULL and How to Pull the raisimLib